sunpy/sunkit-spex #31: Simultaneous Fitting of Spectra from Different detectors/instruments

DanRyanIrish, 2020-03-18:
Lindsay has requested that this be a capability we enable in the long run.

In addition, a fittable scaling factor between different spectra to account for idiosyncrasies of the different instruments.

LinErinG, 2020-03-18:
About the scaling factor - what I have in mind is similar to the way this can be handled in XSPEC.  We use this, for example, in simultaneous fitting to NuSTAR's two telescopes.  It's needed when part of the PSF for one telescope lies over the chip gap, for example, which happens often.  It could also be used when simultaneously fitting to data from two instruments when the data from each don't necessarily agree (e.g. AIA and XRT, for which a scaling factor is often used).  Or it could be used for fitting to multiple RHESSI detectors at a time in the lifetime when there is uncertainty about the relative detector areas.  I'm raising the issue now because it may be something that's easier to bake in from the beginning rather than adding later..

ianan, 2020-03-19:
In XSPEC this works by being able to do simple arithmetic with the models and easily tie the fitting parameters, 
i.e. for detectors A and B your counts models are really

SRM_A # (C_A * f(T_A, EM_A)) and SRM_B # (C_B * f(T_B, EM_B)) 

with variables C_B, T=T_A=T_B, EM=EM_A=EM_B and C_A=1 is fixed, f() is our isothermal model.

rschwartz70, 2020-04-06:
What difference has this scaling factor made to the other spectral parameters? Has this been looked at? There is the case where the detectors are sensitive to different energy ranges, NuStar RHESSI, or the same as with multiple RHESSI detectors. The other question to ask is how stable do you require the factor to be? Suppose you have 30 consecutive time intervals using identical RHESSI detectors over the same energy range.  Make a first cut and obtain scaling factors.  What do you do then? Average them and then refit?  OK, I hear you Iain and Lindsay, let's put in the capability and reserve these other questions for further study.

rschwartz70, 2020-04-06:
We have employed a variant of this capability to balance the RHESSI grids at the insistence of Gordon Hurford and Brian Dennis without any requirement to do the due diligence on verifying the stability of the approach. Hope we do better here.

ianan, 2020-04-06:
Testing has been done on single detector fits vs multiple simultaneously to check that consistent results are obtained and that the scaling parameter is within the expected range (usually about 10% for different combinations of pixels from different modules, or the same module but different sets of pixels from different observations).

The point here is really that we need to make sure our models can be simply combined or scaled to deal with a variety of situations -  i..e not just the about scaling for multiple detectors but say having a model of multiple isothermals - like can be done in ospex or xspec.

rschwartz70, 2020-04-06:
Good work on para 1. I need more explanation on the second paragraph.  I'm confused about what you're saying.  Can you describe this more fully?

LinErinG, 2020-04-06:
Regarding Richard's question about stability -- I think a lot of the uses should need a stable scaling factor (e.g. if the factor is due to something like filter or detector degradation).  However, there are circumstances that might require the scaling factor to change throughout one analysis (like if pointing changes result in the source falling on one detector's chip gap).  However, the latter might have other complications, like needing to modify the response for a different part of the chip.

samaloney, 2022-04-04:
So this can prob be closed after #56 is merged?

closed 2025-02-01 by settwi