sunpy/sunkit-spex #42: Suggestion for package structure

samaloney, 2020-03-25:
Just an idea

```
└── emmission
    ├── non_thermal
    │   ├── bremsstrahlung.py
    │   └── other.py
    └── thermal
        ├── continuum.py
        ├── lines.py
        └── other.py
```

binchensun, 2020-03-25:
Emm, I personally think the first "child" may be better to be categorized in terms of emission mechanisms: bremsstrahlung (free-free), line emission (free-bound, bound-bound), and others (e.g., pion decay). The difference between thermal and nonthermal is only related to the source electron/ion distribution. For example, bremsstrahlung can also take a thermal distribution, and a Kappa distribution bridges both thermal and nonthermal.

DanRyanIrish, 2020-03-25:
@binchensun Are you suggesting that we have one function/model that can produce both thermal/non-thermal bremsstrahlung spectra?  If so, what would the API look like?

binchensun, 2020-03-25:
> @binchensun Are you suggesting that we have one function/model that can produce both thermal/non-thermal bremsstrahlung spectra? If so, what would the API look like?
I have not yet thought about the detailed implementation. But a non-thermal thin-target bremsstrahlung takes in, e.g., a power-law, which normalizes upon n_i*V*n_e_nonthermal (if nonthermal electron distribution is described in density, not in flux), and thermal bremsstrahlung takes in T and EM (the latter has the same unit as n_i*V*n_e_nonthermal). The outputs are both in photons/(cm2 s keV). I guess it should be straightforward to put both the thermal and nonthermal thin-target modules into, e.g., bremsstrahlung.py? Looking at the SSWIDL implementation of thermal bremsstrahlung, it should be pretty straightforward: https://hesperia.gsfc.nasa.gov/ssw/packages/xray/idl/brem_49.pro

DanRyanIrish, 2020-04-02:
So here's an idea for the structure based on the thread here.  (It needs further discussion though.)  Modules/models would be based on emission mechanisms, rather than thermal/non-thermal.  (Although we could also have thermal/non-thermal models that are just wrappers around the relevant mechanisms.)
```
└── emission
    ├── bremsstrahlung.py
    ├── line.py
    ├── free-bound.py
    ├── other.py
    ├── thermal.py  # Wraps around relevant mechanisms and takes thermal distributions as inputs.
    ├── nonthermal.py  # Wraps around relevant mechanisms and takes nonthermal distributions as inputs.
└── particles
    ├── electrons  # Electron distributions
        ├── thermal.py
        └── nonthermal.py
```
We have a module for the particle distributions which are taken as inputs to the emission modules.
Just an idea.

samaloney, 2024-06-14:
So a direction has been chosen in #151  so going to close this issue for now.

closed 2024-06-14 by samaloney