sunpy/sunkit-spex #51: Thick and Think target models updates for a 0.0.1 release

samaloney, 2021-06-17:
Discussion of a the minimal required updates for a release 

- [ ] Add units to inputs, calculation, and outputs see #46
- [ ] Refactor further away from IDL like implementation #21 and #45
- [ ] Documentation / examples

DanRyanIrish, 2021-06-17:
Regarding point 2, I think the bigger thing is to decide on the API.  Refactors for greater speed and/or Pythonic-ness can be done at any stage with little penalty so long as the API doesn't change.  I realise that these two issues are completely separable but if others agree, that will hopefully give more focus.  So the question is, what should the API be?

My initial feeling is that the models should start off as functions.  This is already the case.  Although I'd like to better understand the decision behind making the electron distribution a class.  I'm not suggesting this is not the right thing to do.  But we should understand whether it is the best choice.

samaloney, 2021-07-06:
The electron distribution functions need to be able to return the normalised integral as well as the original distribution and also return the continuous parts where applicable so that why I had them as classes but could work as functions too. It hard to know until someones tires it but something like this maybe

```python
def power_law(x, slope, norm, integral=False):
     if integral is False:
         return x.min(), x.max(), norm * x **slope
     elif integral is True
         return x.min(), x.max(), norm * x ** (slope - 1) # this isn't wright just the idea

def thick_target(photon_edges, edist_func, edist_params=None, observer_distance=1*au, integration_method=scipy.integragate.quad'):
   if not callable(edict_funct):
       raise Type('edist_func must be callable')
   for continuous_part in edict_funct(**edist_params, integral=True)
       lower_limits, upper_limits, values = continuous_part
       res = integration_method(.....)

flux = thick_target(photon_edges, power_law, edist_params={'slope': -5, 'norm': 1000})
```