sunpy/sunkit-spex #55: New name for sunxspex?

hayesla, 2022-04-01:
### Describe the feature

The name is confusing and @samaloney can't remember it

### Proposed solution

new name!

ayshih, 2022-04-01:
Small changes to the name:
* `sunspex` (i.e., drop the interior "x") would address the wavelength issue, but might tie more strongly to SPEX/OSPEX
* `sunspec` would break the connection with SPEX/OSPEX, but might appear to connect to Xspec
* Could also use `sol...` or `solar...` instead of `sun...` if that helps folks with saying it or remembering it

If the complete removal of energy range makes folks nervous, one could use the adjective descriptor of "high-energy" and make a bigger change to the name.  Here are some fun acronyms that I've come up with:
- SASHET: Spectroscopic Analysis Solar High-Energy Tool (pronounced like "sashay")
- SHETSU: Solar High-Energy Total Spectroscopy Utility (pronounced like the dog breed "Shih Tzu")
- PHESS: Package for High-Energy Solar Spectroscopy (somehow pronounced like "fez")

ayshih, 2022-04-01:
For fans of the Princess Bride, how about [PHESSIC](https://princessbride.fandom.com/wiki/Fezzik)?  Package for High-Energy Solar Spectroscopy with Interface and Components...

elastufka, 2022-04-04:
specsun (try saying 'a specsun spectrum' three times fast)

spectools (simple, easiest to remember?)

YASS (Yet Another Solar Spectroscopy package  - besides the allusion to YARN and YAML, also plays tribute to Switzerland's favorite card game)

😬😬

ianan, 2022-04-04:
I would vote for something not *spex or *spec as avoids confusion with other packages

I do like SASHET - a packet of useful things and the chase/dance to get the best fit (i.e. chassé). Acronym a bit awkward, needs a ","? Spectroscopic Analysis, Solar High-Energy Tool.

DanRyanIrish, 2022-04-04:
I do love a good acronym.  But in this case I wonder if it would be better from a branding and discoverability perspective if the package name itself gave some clue to its purpose without already needing to know what the acronym stands for.  This is the case for other Python packages in the solar/astro world (think of sunpy, ndcube, astropy, specutils, etc.)  Perhaps something as simple as **sunspecpy** might be best.  From this we know it's solar (as in SUNpy), relating of spectroscopy (as in astropy SPECutils), and a Python package (PY).  The fact that it ends in "py" might also help break the "spec" link with XSPEC.

A potential downside of sunspecpy is its generality, i.e. not specific to high energy.  But that could also be a strength...?

It's also certainly not as clever a suggestion as those of @ayshih or @elastufka.  But it might be easier for people to find, understand and remember.

ianan, 2022-04-04:
I'd prefer something with 2 syllables, as sunspecpy becomes a bit of a mouthful when trying to use in a sentence.

Note that all of the following already exist for spectral fitting: pyxspec, pyspec, pyspex, spexpy, specpy. And various combinations of sun, sol, spec, spex, and py for other things. There is also SHERPA and BXA.

So my vote is for a short and catchy word.

elastufka, 2022-04-04:
xfit, pronounced "X fit" so as never to be confused with "Crossfit"

pymilo, 2022-04-04:
I'm new, so I might not have much of a vote here. But, for the name discussion I rather like Dan's approach. I did not know of the existence of this package until I used Google to search for "python solar spectral X-rays".

You probably want to have a pretty obvious name that tells users what to expect even before having to click and get to the readme. I love having catchy acronyms for missions (FOXSI, PADRE, COMPLETE, CRAP). But, I have seen that the most successful analysis packages have "boring" and direct (clear) names.

I don't see the three syllable as a problem in python (see, e.g., matplolib or tensorflow). I would emphasize a name that catches senior ssw ospex users, as well as newbies.

ianan, 2022-04-04:
I think my main issue is that with a Scottish accent sun and spec end up with lot's of sh sh sh sh. 

At least with SASHET the sh sh are meant to be there :smile:

ayshih, 2022-04-04:
If the group mostly wants boring and direct/clear, why not just go with `sunkit-spectra`?  This will eventually be very appropriate if the future is to depend on an abstract spectrogram class living in the core `sunpy` library.

> I did not know of the existence of this package until I used Google to search for "python solar spectral X-rays".

To be fair, you didn't know of the existence of this package when it was already named `sunxspex`, so changing the name isn't going to much help/hurt that.  I imagine that once this package actually has a meaty release and gets publicized, the discoverability concern will largely be moot.  What the package is named and what it does will be paired in searchable documentation, presentations, abstracts, and eventually papers.  And, presumably, it will be the only SunPy affiliated package that is used for spectroscopic analysis.

People have managed to find and use `pandas`...

> I love having catchy acronyms for missions

The same motivation applies here for desiring a punchy name that avoids confusion in a crowded namespace with many conceptually similar entries.  Just like you wouldn't want to have every X-ray imager mission to avoid an acronym and try to have a name that is a variation of *XRayImager* (e.g., *ImagerXRays*, *AstroXRayImager*, *SunXRayImager*, etc.), there already are quite a few spectral-fitting packages, as @ianan noted.

The crowded namespace also reinforces the misspelling annoyances with `spex`/`spec`.  It seems better to choose a name that avoids all that.

pymilo, 2022-04-04:
`sunkit-spectra` would get the job done (I like it). Also, following what you and @DanRyanIrish said, not having a name specifically for high energies could also be a strength in the future.

Just to be clear. I knew of the existence of this package when it was already named `sunxspex` (I haven't contributed to it, but I've been an active user; that's why I'm here 🤓). Coming from using ospex, I found this package when looking to see if sunpy had already implemented some of the tools offered by sswidl/ospex.

Regarding pandas, I would argue that even with its current global success, that name is usually problematic for newcomers in data science. Anyone can infer that MatPlotLib is some sort of Library to Plot Mathematical "things". Same with NumPy. With pandas it is different. Almost nobody knows what its accronym is (apparently pandas stands for “Python Data Analysis Library”). I understand that what the package does is typically paired in searchable documentation, and whathaveyou. That allows people to find the tool. Yet, having a direct/clear name would help the package to be even more findable.

I agree with your reasoning of trying to avoid `spex`/`spec`. Having a name like `sunkit-spectra` would help clueless users like me to find the tool quicker 👍🏻

samaloney, 2022-04-04:
I personally like sunkit-spectra it’s different, acknowledges the ties to the sunpy project and is energy range/instrument agnostic. 

I still prefer sunspec but see the issues with this name.

ianan, 2022-04-05:
I would avoid calling it `sunkit-spectra` as for most solar folk (x-ray is sadly the minority) what they want from a spectra package is not what this will provide (certainly not in the short-term anyway). And there is a fundamental difference between x-ray/&gamma;-ray range and what is done/needed a lower energies, so there is a separate debate whether you'd ever want a single "spectra" package that does everything.

So maybe go for `sunkit-spex` as gives us links back to the heritage of OSPEX/SPEX, which will be where most of the existing solar X-ray users will be coming from. Nothing `spec` related as would cause confusion with XSPEC, which is the more minor package for solar folk anyway.

ianan, 2022-04-07:
Still prefer `sunkit-spex`, short and to the point. 

An easy sell of what we have and can do now, and clear focus for coming months/years. 

`from sunkit-spectra import spex` somewhat burys the lead.

ayshih, 2022-04-07:
> sunkit-spex

Basically, you're replacing the middle "x" in `sunxspec` with "kit-".  That solves the not-only-for-X-rays consideration, but I imagine that @samaloney and others will still find themselves misspelling it as `sunkit-spec`. =P

ianan, 2022-04-07:
I'd be happy to remind folk: 🌞 🧰 👓 ❌

elastufka, 2022-04-07:
🤦🏻‍♀️

... unfortunately reaction emoji don't yet include The Facepalm

ianan, 2022-06-13:
Do we want to get the name changed sorted before the SPHERE meeting? 

My vote would still be for `sunkit-spex`

hayesla, 2022-06-13:
yes it'll be good to have it changed by then.

I'm personally still fond of `sunkit-spex` too

samaloney, 2022-06-14:
What is the 'spex' bit short for or acronym of?

ianan, 2022-06-14:
SPectral EXecutive

https://hesperia.gsfc.nasa.gov/ssw/packages/spex/doc/ospex_explanation.htm#OSPEX%20Overview

samaloney, 2022-06-21:
Ok I dumped all the suggestions into the poll below, I don't know if voting is the best way forward here but if there's a clear favourite it might be easy 🤷 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ianan, 2022-06-21:
You might want to google some of those suggestions before making them voting options - that could discount the PHESS* ones....

ayshih, 2022-06-21:
> Please do not use `sunkit-spectra`

See [my thoughts above](https://github.com/sunpy/sunxspex/issues/55#issuecomment-1088864801)

Cadair, 2022-06-21:
I am happy for this to become sunkit-spectra as long as what the scope is and how the subpackage layout works is clearly documented somewhere :smile:

samaloney, 2022-07-01:
Ok so `sunkit-spex` was the "winner" with 8 votes 😆 unless there's any strong objection I think we should proceed with this at least for the moment.

ayshih, 2022-10-05:
So...  Is this name change going to happen?

ianan, 2022-10-06:
Good question @ayshih - from talking to @samaloney at SO8 I think some minor changes/updates were in the pipeline and then could name change and zenodo to get a doi for those publishing with the current version? Then next version would start/do the greater internal overhaul so that has at least same functionality but also have a more solid/logical infrastructure to build upon?

`sunkit-spex` is better than '`sunxspex` but still not great....