sunpy/sunkit-spex #83: Add STIX spectrogram, background FITS, and proper SRM loading

settwi, 2022-10-04:
### Describe the feature

STIX data come in a variety of formats. One appealing format is spectrogram data. It is downlinked for every flare, provides good energy and time resolution, and can be readily used for spectroscopy.

I noticed sunxspex doesn't have a way to load in spectrogram data natively, nor does it support easy background-inclusion using the STIX background fits files. It also expects the SRM as a strange file format.

I propose adding:
- The ability to load and process STIX spectrogram data
- The ability to load and process STIX background FITS files and use them in conjunction with other science data
- The ability to load different SRMs for different attenuator states

### Proposed solution

I have implemented a version of a spectrogram loader called `StixSpectrogramLoader` that reads the L1A spectrogram FITS, a SRM FITS output by IDL, and a background FITS and parses them into the correct sunxspex format. The supporting code is included in the attached gzip file.

There is a notebook demonstrating how it all works by investigating a flare which was analyzed in [Massa et al. 2022](https://link.springer.com/article/10.1007/s11207-022-02029-x). The parameters that sunxspex finds agree with the published parameters within uncertainties.

Anything not covered by the sunxspex license in the code I've uploaded can be considered to be covered by The Unlicense.

[stix-sunxspex.tar.gz](https://github.com/sunpy/sunxspex/files/9710726/stix-sunxspex.tar.gz)

I propose that bits of this code be adopted into the sunxspex codebase to allow for the analysis of STIX spectrogram data.

ayshih, 2022-10-05:
Apologies for this tangent, but:

> Anything not covered by the sunxspex license in the code I've uploaded can be considered to be covered by The Unlicense.

There are the concerns that the Unlicense would not function as intended in some jurisdictions, and your phrasing makes me additionally concerned that there isn't a bright line about which code is covered by each license.  Assuming that you are the author of the code that is licensed under the Unlicense, I recommend that you dual license the code so that all of the code can be treated as being under the `sunxspex` license.

settwi, 2023-06-05:
Now that Issue #97 and [PR #98](https://github.com/sunpy/sunxspex/pull/98) have been brought, I think it's time we adapt the code I wrote here into a proper sunxspex format.

I have a fork over here that can be a starting point: [settwi/sunxspex](https://github.com/settwi/sunxspex). I also have some more up-to-date code locally that we can look at, maybe in a `sunxspex` meeting in the coming days.

@KriSun95 @natsat0919

Thanks!

closed 2023-07-10 by settwi