sunpy/sunkit-spex #107: Unable to pass instrument loader classes to the main fitting class directly

KriSun95, 2023-06-27:
### Describe the bug

# Bug

The main fitting class [`sunxspex.sunxspex_fitting.fitter.SunXspex`](https://github.com/sunpy/sunxspex/blob/master/sunxspex/sunxspex_fitting/fitter.py#L62) cannot take an instrument loader class directly because of a [bug](https://github.com/sunpy/sunxspex/blob/master/sunxspex/sunxspex_fitting/data_loader.py#L143) in the [`sunxspex.sunxspex_fitting.data_loader.LoadSpec`](https://github.com/sunpy/sunxspex/blob/master/sunxspex/sunxspex_fitting/data_loader.py#L20).

The bug causes an error when passing a user made instrument loader since I've written to check the user class is a subclass, instead of just an instance, of the [`instruments.InstrumentBlueprint`](https://github.com/sunpy/sunxspex/blob/master/sunxspex/sunxspex_fitting/instruments.py#L30) class. The current bug makes it difficult to troubleshoot current instrument loaders and create new ones.

The fitting class *should* be able to take the instrument loaders and future updates/discussions might make this the preferred way to pass information to the fitting class. I.e., rather than passing different instrument data files to the fitting class directly the files should be handed to the relevant instrument loader first then the instance of the loader passed to the fitting class.

# Proposed Solution

Although more work will be done on the loading side of this, fixing this seems to be a good thing to do. I will open a PR with my solution of replacing the incorrect line with one that only checks if the instrument class being passed is an *instance* of `instruments.InstrumentBlueprint`. Tests will also be written to ensure expected behaviour.

### To Reproduce

```
from sunxspex.sunxspex_fitting.instruments import NustarLoader
from sunxspex.sunxspex_fitting.fitter import SunXspex

# could be any original instrument class, not just NustarLoader()
class nu(NustarLoader):
    def __init__(self, *args, **kwargs):
        NustarLoader.__init__(self,*args, **kwargs)

# give my new instrument loader my specific data
d_dir = "/Users/kris/Documents/umnPostdoc/projects/analysis/nustarNov2021/nestedSampling/analysis/ns_products/"
nu_inh = nu(pha_file=d_dir+'evt_2112_6to10keV/nu20618003001A06_cl_grade0_sr.pha')

# pass the new instrument loader instance to the main fitting class
f = SunXspex(nu_inh)
```

### Screenshots

[image]

settwi, 2023-07-10:
If this works I say just merge it...

KriSun95, 2023-07-11:
As I have suggested here and @settwi has explicitly stated in the PR, once this bug gets fixed, I think it would be better and cleaner to remove the choice of passing files directly to the fitter and only support passing the instrument loader classes.

settwi, 2023-09-23:
Can we close this issue?

KriSun95, 2023-10-02:
This Issue has been addressed in #108 and discussion on passing instrument loaders instead of files and dictionaries has moved to #119.

closed 2023-10-02 by KriSun95