sunpy/sunkit-spex #125: Using some `float64` values with `bremsstrahlung_thick_target` returns all NaNs

KriSun95, 2023-10-18:
### Describe the bug

Issue when using `numpy.float64` numbers for the parameters in `sunxspex.emission.bremsstrahlung_thick_target` as it ends up returning all `nans` and `infs`. Weirdly, rounding to 15 decimal places with the values that return `nans` and `infs` fixes this.

I've "fixed" this in `sunxspex.sunxspex_fitting.photon_models_for_fitting.thick_fn` by taking the inputs and rounding to 15 decimal places via `np.round([total_eflux, index, e_c], 15)` but obviously this isn't ideal.

I will come back and add more detail when I find and log values that produce this weird behaviour but wanted it mentioned somewhere so it doesn't get lost.

samaloney, 2025-04-23:
Think this is solved in #199?

settwi, 2025-08-28:
@samaloney yes

closed 2025-08-28 by settwi