sunpy/sunkit-spex #127: sunkit_spex logo?

ianan, 2023-11-07:
### Describe the feature

As @KriSun95 mentioned the python package folk love a logo - so someone should make one for the newly christened sunkit_spex.

### Proposed solution

Someone make a logo.

settwi, 2023-11-07:
Thanks for the proposed solution. I wouldn't have thought of that...

ianan, 2023-11-07:
Well we could get "ai" to do it for us instead, i.e.
[image]

ianan, 2023-11-07:
or via another image creator - although this might be more related to the band.
[image]

KriSun95, 2023-11-07:
Those AI logos don't look the worst actually, especially the design of the second one. I tried that a while ago and just got back swirling messes with the ones I was using. 

I did end up spending too much time one night working on something...
[image]
...all the professional graphic designers use PowerPoint as well, right? Not sure if it's the one to go with but I had fun making it!