sunpy/sunkit-spex #128: Add albedo correction

samaloney, 2023-11-09:
### Describe the feature

A few people have been asking if we have implemented the albedo correction similar to OSPEX

settwi, 2023-11-09:
I think that would be an excellent addition. I believe OSPEX implements it as a “pseudo-model” but it shouldn’t be too hard to add if we look up the proper equations.

KriSun95, 2023-11-10:
This has been on my to-do list for a while, I've just never been able to get around to it properly. 

I _think_ OSPEX does it via altering the response matrix before the fit and I _think_ the main files to look at are [drm_albedo.pro](https://www.astro.gla.ac.uk/users/eduard/rhessi/albedo/kim/drm_albedo.pro) and [drm_correct_albedo.pro](https://www.astro.gla.ac.uk/users/eduard/rhessi/albedo/kim/drm_correct_albedo.pro) (found [here](https://www.astro.gla.ac.uk/users/eduard/rhessi/albedo/) with some description).

Instructions on how the user interacts with this in OSPEX and some history is [here](https://hesperia.gsfc.nasa.gov/ssw/packages/spex/doc/ospex_explanation.htm#Albedo%20Correction).

samaloney, 2023-11-10:
Yea those sav files are nightmare fuel do we have any idea how they were generated? @KriSun95 I think you are correct about the current implementation/approach here's the relevant [paper](https://arxiv.org/pdf/astro-ph/0510167.pdf) but not clear if that the best way or have a model chain `physical-model->albedo-model->SRM`

edkontar, 2024-09-10:
Green function matrices are now available in JSON format (also as IDL SAVE files) [https://github.com/edkontar/albedo](https://github.com/edkontar/albedo)

settwi, 2024-09-11:
@edkontar thanks, Eduard! this will be very useful for albedo corrections going forward.

would you like to help implement the corrections in Python?

edkontar, 2024-09-11:
> would you like to help implement the corrections in Python?

Yes, I would be happy to help

samaloney, 2024-09-12:
I have a draft PR open which uses the sav files from SSW in #161

What we eventually need is the ability to generate the "green's matrices"

natsat0919, 2024-11-14:
I'm still a bit confused by the albedo correction method. If I understand correctly, in the OSPEX, the albedo correction directly modifies the SRM but in our case, we will generate the photon models -> fold it though the albedo correction matrix-> fold this through the SRM to get the count model. During the fitting process, this method is done to each photon model. That means that all our output best fit count models by default have albedo accounted for them during fitting. But OSPEX always outputs the albedo correction as a separate model component during plotting (with the other models e.g. thermal, non-thermal.. plotted as well). Does that mean that during fitting we are account for albedo within each photon model, but then during plotting we plot the count models without the albedo correction (so in our case just : get best-fit the photon models -> fold through SRM) and instead, we plot a composite albedo model? Or am I misunderstanding this?

edkontar, 2024-11-19:
Prior to April 2010, the albedo correction was applied to the DRM once when the DRM file was read, and the modified DRM was stored for use during analysis (this is how it was handled in old SPEX).  This proved to be a problem for two reasons. 1. When using the drm_mod function component, the albedo correction was lost when the DRM was recalculated in the drm_mod procedure.  2. In May 2010, albedo was added as a function component to allow you to fit the anisotropy parameter:   [pseudo function](https://hesperia.gsfc.nasa.gov/ssw/packages/spex/doc/ospex_explanation.htm#Albedo%20Correction)

Assuming isotropic X-rays: for a given photon spectrum I(eps), albedo corrected spectrum is I_corrected =(1+A)##I, and the count spectrum C_corrected=SRM#I_corrected

edkontar, 2024-11-19:
After doing a simple thermal+thick+albedo fit, the results in photon space should look like this:
[image]

edkontar, 2024-11-22:
I have found IDL code used in 2005 to generate Green Matrices for OSPEX and put it on github:

https://github.com/edkontar/albedo

samaloney, 2024-11-22:
Yea I think the main concept as Eduard said is the we are interested in $I_{emmited}$ but what we model/fit is $I_{obs} = I_{emmited} + I_{albedo}$ so in OSPEX this is 'hidden' in the DRM. I think in our case we want to be explicit and show as model component, even if it is often with fixed parameters ($\theta$, anisotropy). If we want to extract some parameters e.g. from a warm_target model, they can just be extracted from that model components parameters unless I'm missing something?

PR #161 (2025-02-05): Add albedo correction
## PR Description

A a function to read in precomputed greens table and return spectrum with albedo correction, would close #128

- [x] Split function at the moment every time the function is called we would make web request and load a file from disk both very slow not ideal for fitting.   The idea would be split up into at least two parts depending on the use case/s. As far as I know often $\theta$ is fix so the loading and interpolation between $\theta$ can be cached. So one function to load the appropriate greens matrix file and interpolate for given angle. A second fu [... cut]
closed 2025-02-05 by samaloney