sunpy/sunkit-spex #135: Fitting custom spectrum with energy range below 1 keV with thermal (f_vth) model.

abhilash-sw, 2024-01-31:
### Describe the bug

I am trying to fit a custom spectrum which has spectral data below 1 keV (though not usable). I am trying to fit the spectra with above 1 keV using `spec.energy_fitting_range` attribute. 

I am getting "ValueError:  All input energy values must be within the range 1.0002920302956426--200.15819869050395 keV."

Subsequently I tried to limit my input spectrum above 1 keV. I am faced with IndexError (full error attached below.)

### To Reproduce

spec_data_dict = {'count_channel_bins':channel_bins, 'counts': counts, 'count_error':counts_err, 'effective_exposure':exposure, 'srm': rsp_mat}

spec = Fitter(spec_data_dict)

spec.loglikelihood="poisson"
spec.model = "C*f_vth"

fiter=[1.83,15.0]

spec.energy_fitting_range = fiter
spec.params["C_spectrum1"] = {"Status":"frozen"}

spec.params["T1_spectrum1"] = {"Value":10.0, "Bounds":(5.0, 25.0)} #Plasma temperature in megakelvin
spec.params["EM1_spectrum1"] = {"Value":1e-1, "Bounds":(0.01, 1.0)} #Emission measure in units of 1e46 cm^-3.

print(spec.params)

minimised_params = spec.fit()

### Screenshots

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[41], line 1
----> 1 minimised_params = spec.fit()

File ~/anaconda3/lib/python3.10/site-packages/sunkit_spex-0.1.dev156+gcfd41b9-py3.10.egg/sunkit_spex/sunxspex_fitting/fitter.py:2135, in Fitter.fit(self, **kwargs)
   2127 kwargs.pop("_abs_hess_step", None)
   2129 # this has been replaced by a `_run_minimiser_core` so that this can be swapped out easily down the line
   2130 # soltn = minimize(self._fit_stat_minimize,
   2131 #                  free_params_list,
   2132 #                  args=stat_args,
   2133 #                  bounds=free_bounds,
   2134 #                  **kwargs)
-> 2135 soltn = self._run_minimiser_core(minimise_func=self._fit_stat_minimize,
   2136                                  free_parameter_list=free_params_list,
   2137                                  statistic_args=stat_args,
   2138                                  free_param_bounds=free_bounds,
   2139                                  **kwargs)
   2141 self._minimize_solution = soltn
   2143 std_err = self._calc_minimize_error(stat_args, step=_hess_step, _abs_step=_abs_hess_step)

File ~/anaconda3/lib/python3.10/site-packages/sunkit_spex-0.1.dev156+gcfd41b9-py3.10.egg/sunkit_spex/sunxspex_fitting/fitter.py:2097, in Fitter._run_minimiser_core(self, minimise_func, free_parameter_list, statistic_args, free_param_bounds, **kwargs)
   2063 def _run_minimiser_core(self, minimise_func, free_parameter_list, statistic_args, free_param_bounds, **kwargs):
   2064     """ Allows user (or us) to define their own (different) minimiser easily.
   2065
   2066     This should return the same type of output as Scipy's minimize at the minute. This just passes
   (...)
   2095     the parameter results in the order of `free_parameter_list`.
   2096     """
-> 2097     return minimize(minimise_func, free_parameter_list, args=statistic_args, bounds=free_param_bounds, **kwargs)

File ~/anaconda3/lib/python3.10/site-packages/scipy/optimize/_minimize.py:684, in minimize(fun, x0, args, method, jac, hess, hessp, bounds, constraints, tol, callback, options)
    681     bounds = standardize_bounds(bounds, x0, meth)
    683 if meth == 'nelder-mead':
--> 684     res = _minimize_neldermead(fun, x0, args, callback, bounds=bounds,
    685                                **options)
    686 elif meth == 'powell':
    687     res = _minimize_powell(fun, x0, args, callback, bounds, **options)

[... 95 lines cut ...]
    442 # Use binned_statistic to determine which spectral bins contain
    443 # components of line emission and sum over those line components
    444 # to get the total emission is each spectral bin.
    445 flux = stats.binned_statistic(line_spectrum_bins, split_line_intensities,
    446                               "sum", n_energy_bins, (0, n_energy_bins-1)).statistic

File ~/anaconda3/lib/python3.10/site-packages/sunkit_spex-0.1.dev156+gcfd41b9-py3.10.egg/sunkit_spex/thermal.py:626, in _weight_emission_bins_to_line_centroid(line_peaks_keV, energy_edges_keV, line_intensities)
    624 new_iline = copy.deepcopy(iline)
    625 if len(neg_deviation_indices) > 0:
--> 626     neg_line_intensities, neg_neighbor_intensities, neg_neighbor_iline = _weight_emission_bins(
    627         line_deviations_keV, neg_deviation_indices,
    628         energy_center_diffs, line_intensities, iline, negative_deviations=True)
    629     # Combine new line and neighboring bin intensities and indices into common arrays.
    630     new_line_intensities[:, neg_deviation_indices] = neg_line_intensities

File ~/anaconda3/lib/python3.10/site-packages/sunkit_spex-0.1.dev156+gcfd41b9-py3.10.egg/sunkit_spex/thermal.py:677, in _weight_emission_bins(line_deviations_keV, deviation_indices, energy_center_diffs, line_intensities, iline, negative_deviations)
    672     b = 1
    674 # Calculate difference between line energy and the spectrum bin center as a
    675 # fraction of the distance between the spectrum bin center and the
    676 # center of the nearest neighboring bin.
--> 677 wghts = np.absolute(line_deviations_keV[deviation_indices]) / energy_center_diffs[iline[deviation_indices+a]]
    678 # Tile/replicate wghts through the other dimension of line_intensities.
    679 wghts = np.tile(wghts, tuple([line_intensities.shape[0]] + [1] * wghts.ndim))

IndexError: index 58 is out of bounds for axis 0 with size 58

ianan, 2024-02-14:
The lookup files that the thermal model uses were originally created for ospex in sswidl using CHIANTI and cover 1-250 keV - they are the v71 files here https://hesperia.gsfc.nasa.gov/ssw/packages/xray/dbase/chianti/

So you would have to use other files, that directory contains v901 ones that go down to either 0.1 or 0.7 keV (01_12 or 07_12 files), though there are still other issues at the moment - see issue #99 for more info.

ianan, 2024-03-05:
I've generated new files that go down to 0.1 keV and can be found here: https://gla-my.sharepoint.com/personal/iain_hannah_glasgow_ac_uk/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fiain%5Fhannah%5Fglasgow%5Fac%5Fuk%2FDocuments%2Fchxdb&ga=1

I tested them in sunkit-spex and seem to be fine (though takes a while to load) https://github.com/ianan/fvth_stuff/blob/main/better_chxdb/check_01files.ipynb

[image]