sunpy/sunkit-spex #185: There are two thermal modules

settwi, 2025-01-31:
### Describe the bug

There are two thermal modules, one in `sunkit_spex.legacy.thermal` and one in `sunkit_spex.models.physical.thermal`.

Recently in #178 we patched some issues with the thermal emission continuum but only in the `legacy` module. So, things are starting to diverge between the two modules.

I guess, if things change in the new thermal emission version, it isn't really an issue because the legacy version will be independent. But, for now, if people want to use the thermal emission, it's unclear which should be used.

Should we move the `legacy` emission to the `models.physical` model and then just have the `legacy` version import the current one? Or, should it be the other way around?

settwi, 2025-02-04:
What if we just import legacy in the new module?

samaloney, 2025-02-05:
But then if we need to change the thermal code for example to better integrate the astropy modelling API we have to update other that uses it in legacy. The way I look at it legacy basically a read only copy bar bug fixes.

closed 2025-02-06 by settwi