sunpy/sunkit-spex #220: Make a development guide

settwi, 2025-07-05:
### Provide a general description of the issue or problem.

We need a guide that says how to initialize a dev environment and run it properly. It could just be a markdown file on GitHub; it doesn't need to necessarily be an offical document that we put into Sphinx. It would be good to link to relevant documentation as well in case something breaks in the future.

There are several tools used to check the code:
- tox (which manages the rest of them)
- ruff
- buildthedocs/sphinx
- isort
- codecov
- GH actions & their configurations
- pytest
- ...?

It's not immediately clear what all is required. For instance
- do we need isort if we have ruff?
- Should we have a `setup.py` when we have a `pyproject.toml`?

**Let's document how things work, decide what we need/don't need, then remove what we don't?**

settwi, 2025-07-07:
Along these lines, should we bump the minimum required version here?
https://github.com/sunpy/sunkit-spex/blob/59a64729cce4b4bfe68eb79a4458915a9b7568d0/.sunpy-template.yml#L11

Are there other spots where a minimum version is written?

Cadair, 2025-11-17:
> Still, I'm wondering, do we need isort if we have ruff, for instance?

This is due to ruff lacking features in it's isort version which made the migration more painful than we wanted it to be see: https://github.com/sunpy/package-template/issues/230

> Also, it might be nice to have a more fleshed-out index page rather than just the headers generated by Sphinx, like a paragraph or section of bullet points describing where to go if you are curious about different parts of the packaging process.

Contributions to improve the package template documentation are very welcome! This sounds like a great improvement, and I'd be happy for it to live on the package template index page.

> Along these lines, should we bump the minimum required version here?

This file should just be deleted, it's not used by the new template any more.